jasbur/RaspiWiFi #4: Gpio reset not working on Pi Zero W

cwking85, 2017-05-09:
Love the program and it works great on my Pi Zero W with the exception of the gpio reset. I modified  reset.py to use gpiozero instead of RPi.GPIO and it solved my issue.

jasbur, 2017-05-21:
Thanks for the heads up. I finally found a Pi Zero W and it's on its way. I'll look into this as soon as I get it.

jasbur, 2017-09-01:
I finally got ahold of a Raspberry Pi zero W and tested the reset function. I applied 3.3 volts from pin 1 to GPIO18 (physical pin 12) and it reset just fine for me with the existing code. Can you think of anything unique about your setup that might have caused it to act differently?

jasbur, 2017-09-01:
I spoke too soon. The reset happened but the original configuration files don't seem to have been copied. Looking into it.

jasbur, 2017-10-17:
This seems to have been a pathing issue and should be resolved by recent updates.

closed 2017-10-17 by jasbur