jasbur/RaspiWiFi #7: RPi Zero W issue ( unable to connect to the AP{RaspiWifi} )

akshaykulkarni46, 2017-08-31:
I tried this on RPi Zero W and I was able to create the AP {RaspiWifi} but unfortunately i was unable to connect to that AP using mobile or PC

jasbur, 2017-08-31:
Can you provide any more information, such as which devices you tried to connect with or any errors messages you received?

akshaykulkarni46, 2017-08-31:
Hello Sir , 
I tried this on Raspberry pi Zero W and while installing Rails gems using sudo bundle install I had some issues . 
Thanks & Regards
- Akshay Kulkarni
  IQ Root Technologies ,
  Bengaluru  -560056

---- On Thu, 31 Aug 2017 22:20:55 +0530 notifications@github.com wrote ----

Can you provide any more information, such as which devices you tried to connect with or any errors messages you received?

adilberrada, 2017-09-18:
Hi thank you for your code ! I am having the same issue as @akshaykulkarni46. I am unable to connect to the AP using my mobile or PC. I am using a rpi3 and tried to connect with my mac and iPhone.

Any idea ?
Thanks a lot

jasbur, 2017-10-17:
@adilberrada Recent updates seem to have resolved the issue for @akshaykulkarni46 Have you had a chance to try a more recent version of the code?

closed 2017-11-05 by jasbur