jasbur/RaspiWiFi #10: After Install and Reboot, I Enter My Password and it Reboots Again Without Being Connected to the WiFi That I Just Put The Password In For

smallzZz8, 2017-10-16:
So I am running Raspbian Stretch with Desktop (Latest Version) with my RPi 3 model B+. I installed everything and it seems to work. Once it finishes its first reboot I see the device under my phones Wifi selections and I select it. Then I navigate to 10.0.0.1 and enter the password. The Pi then reboots. Once it is back on, its not connected to the internet? What could it be? Its almost working lol.

smallzZz8, 2017-10-16:
I know that isn’t much to work off of but I do remember that the bundle install took quite some time, so maybe that is the issue? 

Any commands that I can run to make sure things are working correctly?

jasbur, 2017-10-16:
Its not unusual for the bundle install to take a bit of time, especially on a B+. A couple of the gems that install need to compile native extensions from source and that's just CPU intensive. Besides, any error in that part of the setup would result in the web interface not coming up for you and it sounds like that part worked.

My guess is that the wpa_supplicant.conf didn't get edited or copied correctly. Can you run:

`cat /etc/wpa_supplicant/wpa_supplicant.conf`

and copy and paste the output here for me?

smallzZz8, 2017-10-16:
First off huge thanks for the reply! I stayed up all night trying  I get it to work, but was unable too. I’m still new to the rpi community but I’m learning quick. 

Unfortunately Mondays are my busy days and I just left the house and won’t be back until about 8ish. But the first thing I will do when I get home is run that command. I will be on all night as well so I’ll be responding fast. 

Again thank you for the reply and I’ll be reply tonight for sure!

smallzZz8, 2017-10-16:
Okay as promised here it is. I replaced my real password with something else. The ssid and password to be correct. 

```
pi@raspberrypi:~ $ cat /etc/wpa_supplicant/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
	ssid="Home57"
	psk="MYWIFIPASSWORD"
	proto=WPA2
	key_mgmt=WPA-PSK
}
pi@raspberrypi:~ $ 

```

smallzZz8, 2017-10-17:
Ya I just used another sd card and reinstalled the latest version of Stretch with Desktop on my pi. Same outcome and same `cat /etc/wpa_supplicant/wpa_supplicant.conf` contents.

jasbur, 2017-10-17:
Well, that `wpa_supplicant.conf` seems to be correct so we know the configuration app seems to have done its job correctly. Have you tired pinging from the Pi to see if you have any access at all? 

`ping google.com -c 10`

What's the last block of output there? Also, can you double check the information in the `wpa_supplicant.conf` against your actual network information? Capitalization and spaces do matter for both the SSID and WPA Key.

smallzZz8, 2017-10-17:
Ok so I ran the ping and I got this:

```
pi@raspberrypi:~ $ ping google.com -c 10
ping: google.com: Temporary failure in name resolution
```

And yes the spelling and capitalization is correct on the SSID and password. Here is the ifconfig:

```
pi@raspberrypi:~ $ ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 4  bytes 156 (156.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4  bytes 156 (156.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.1  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::ba27:ebff:fee3:a39f  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:e3:a3:9f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 34  bytes 5058 (4.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@raspberrypi:~ $ 
```
I am not sure if it is suppose to change but the wifi bars on the top right still remain with X's. Thanks for the replies.

smallzZz8, 2017-10-17:
A quick update to my last post to try to make this easier. I have several SD cards laying around and I wanted to give you my installation log so I took another SD card and installed Raspbian Stretch with Desktop the most latest release and I got the same exact problem. 

The Pi gives off Wifi, I enter credentials, Pi reboots, still not connected to the internet, I got the same ping as above, same `ifconfig `as the one above, and same `wpa_supplicant.conf` as above. After I enter the credentials and the Pi reboots, the Pi is still giving off a wifi so maybe that is the reason it is not connecting to my wifi because it is giving off wifi instead. Idk just a thought.? 

But I wanted to give you a full log of the installation but I think it is a bit too long here for this thread so check out this Pastebin. https://pastebin.com/0Cm4yisa If you look closely, when it is installing the `isc-dhcp-server` there is a fail (just ctrl+F and search fail to see what I mean). Could that be why? Anyways hope this helps a little bit more.

jasbur, 2017-10-17:
I'm on my phone at the moment, so I'll look closer later on. But right now it seems to me like maybe the interfaces file didn't copyright during the setup. That basically switches it from a host to a network client. Try copying it manually and see if that has any effect. You'll have to run the following command from within the RaspiWiFi directory.

`sudo cp -r Reset\ Device/static_files/interfaces.apclient /etc/network/interfaces`

smallzZz8, 2017-10-17:
Ok I ran the command after `cd RaspiWiFi`, then rebooted and still not connected. However I am now realizing that the Pi is no longer giving off its own wifi.

jasbur, 2017-10-17:
What does the output of `ifconfig` look like now?

smallzZz8, 2017-10-17:
It looks the same as the one above. I am not home at the moment but I typed that command in before I left my house. I should have posted it here. I can later when I get to my house if needed.

jasbur, 2017-10-17:
Yeah, if you're able to. I'll also get a change to test it out on my Pi 3 tomorrow. If I encounter the same issue you won't have to keep feeding me info then.

smallzZz8, 2017-10-17:
Ya when I get home ill definitely paste in my ifconfig. Ill be on until real late tonight in case you plan to be as well. Again thanks for helping man!

smallzZz8, 2017-10-18:
Ehh sorry didnt expect to take that long to get home. Here is the `ifconfig` after running `sudo cp -r Reset\ Device/static_files/interfaces.apclient /etc/network/interfaces` and rebooting.

jasbur, 2017-10-18:
Everything looks good in the `ifconfig` output this time. It just occurred to me that i developed RaspiWiFi on Raspbian Jessie though. I actually didn't know the images had been updated to Stretch (even though I know you mentioned it right away).

I'm going to start testing on Stretch this afternoon. In the meantime, I bet if you try using an image based on Jessie it will work.

You can download the latest Jessie-based image here:

http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/2017-07-05-raspbian-jessie.zip

or the torrent for it here:

http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/2017-07-05-raspbian-jessie.zip.torrent

smallzZz8, 2017-10-18:
Awesome, I’ll try when I get home later. I will let you know how it goes.

smallzZz8, 2017-10-18:
Ya still didnt work with that version of Jessie which doesnt make sense to me. Either the code must be messed up, my pi must be messed up or my router is not allowing it to connect. Interested to see if it works for you.

jasbur, 2017-10-19:
I've reproduced the problem under Stretch on my end. I'll let you know as soon as I figure something out.

jasbur, 2017-10-19:
I might have found the problem. Apparently some network configuration files were changed in Stretch. Try cloning the `v1.3_beta` branch and see if that works.

Make sure to start fresh with a newly imaged Stretch installation.

smallzZz8, 2017-10-19:
It works! Thanks a lot man, now I can continue the rest of my project!

jasbur, 2017-10-19:
Awesome! We did it! Thanks for all the help and feedback. I was only able to find the offending line of code with all of the information you provided. Good luck with your project.

closed 2017-10-19 by jasbur