jasbur/RaspiWiFi #11: Changing network

smallzZz8, 2017-10-20:
Ya I know, this guy is back again Lol. So I've come across an issue.

So lets say you are at house 1. You start the pi and everything works perfect. But now lets say you go to house 2. You power on the pi expecting the pi to give off wifi so you can enter the password credentials. Well, the pi does give off wifi, however not under the 10.0.0.1 server. 

So I thought of maybe a quick fix to this. What if the pi on startup asked itself, do I have wifi? If no then run the `sudo python3 initial_setup.py` command. If yes than do nothing. Thoughts?

To try this for yourself just unplug your wifi instead of going to another house. Since your wifi is unplugged then it should make the 10.0.0.1 server. But if it does then we know it is working.

jasbur, 2017-10-23:
Actually that's why  I have GPIO pin 18 set up to receive a button signal. If you have a button hooked up to it and press it for 10 seconds straight it will reboot itself back into the configuration mode. 

I would certainly be possible to have that reset trigger automatically if it doesn't see the previously configured access point, but I can imagine some situations where you wouldn't want it to do that. For instance if you setup a device while at home and feed it all the information it needs to be used off-site then brought back to its home environment for more configuration.

Thoughts?

closed 2017-11-05 by jasbur

lvmasterrj, 2023-05-04:
Sorry for re-oppening this topic. I was looking for this exact behavior of RaspiWifi to reset when the Raspberry Pi turns on and doesn't connect to a network and ended up here, where I find out it wasn't it's default behavior.

To achieve what I was looking for, I edited my rPi "/etc/rc.local" and added the following script:

`_IP=$(hostname -I) || true

if [ "$_IP" ]; then
   printf "My IP address is %s\n" "$_IP"
else
   printf "No network detected. Reseting RaspiWifi..."
   sudo python3 /usr/lib/raspiwifi/reset_device/manual_reset.py`

This way, I can go to someone's house with my rPi and when it doesn't detect the wifi it will auto reset the RaspiWifi for me to enter a new Wifi.

Hope this helps someone.