jasbur/RaspiWiFi #17: DHCP Issue - 169.x.x.x address

lukewis, 2017-12-14:
Been having an issue (seen on Pi Zero W and Pi B+) where the pi fails to assign itself an ip address.  It broadcasts the RaspiWifi network, but it lists the ip address as a 169.x.x.x address instead of the desired 10.0.0.1.  The Pi B+ eventually seemed to get an ip address (after many reboots), but my Pi Zero seems stuck in this state.  Any thoughts or items to check?

lukewis, 2017-12-15:
I'm thinking this is a bug in the Raspian GUI......when I hover over the wifi indicator in the status bar, it shows the 169 address, but I can indeed connect to 10.0.0.1.  I also see that the 10.0.0.1 address is set statically in /etc/network/interfaces, so it's not being assigned by DHCP.

closed 2017-12-15 by lukewis

jasbur, 2018-03-23:
I just released version 1.5 that should fix this issue.