jasbur/RaspiWiFi #18: Breaks networking entirely

bottleworks, 2018-01-14:
Hi. 
I'm having an issue using this.  On fully stock Raspbian Stretch install, this completely breaks the networking.  I see it's modifying /etc/network/interfaces.  I understand Raspbian has moved away from using this method of configuration.  I suspect this might be the cause?

I've also noticed that initial_setup.py doesn't install RPi.GPIO.  
That's all I got for now.  Trying to figure it all out...

bottleworks, 2018-01-15:
It looks like nodejs also needs to be added to the setup script.

jasbur, 2018-01-15:
Actually, I have a condition I'll check when setting to AP client mode to not use the `/etc/interfaces` file when running stretch. That happens in the `main.rb` file here:

`Configuration App/app/models/main.rb`

So, I'll have to look into whatever is going on there. You're right about NodeJS though. I'll make sure to add it to the Install script.

closed 2018-01-19 by bottleworks

pyrosmiley, 2018-03-07:
@bottleworks I was able to get your fork running (but not the original) on a Pi zero w, but just a heads-up: the SSID of the pi's network was showing as "WifiNetworkName"

bottleworks, 2018-03-07:
Thanks.  I've updated the readme file and corrected a couple grammar issues.  It's a good thing I wrote everything out there.  I'm already starting to forget the mechanics of the system!

jasbur, 2018-03-23:
@bottleworks I just released version 1.5 that should most/all of the issues you have been having.