jasbur/RaspiWiFi #19: Only  see nginx

marcospolanco, 2018-02-19:
Upon installation and reboot on a Raspberry Pi 3, only the default nginx index.html page is visible when connecting to 10.0.0.1. In terms of process, I re-installed several times...the program stopped with errors so I explicitly installed the requisite pages as instructed:

`sudo gem install jason -v '1.8.2'`
`sudo gem install binding_of_caller -v '0.7.2'`

The last run of initial.setup.py ran without a hitch. On a related note, since the configuration page is not appearing, how might I reconfigure the Rpi to connect to WiFi, short of re-imaging it, should the RaspiWifi page not appear. Thanks.

jasbur, 2018-02-21:
I've just reinstalled myself on a clean version of raspbian stretch and had
no issues. I didn't have to manually install any packages or anything, I
just ran the initial configuration script.

Nginx doesn't seem to be present on my base install of raspbian. Being that
it's also a web server it's most likely expecting a connection on Port 80
which is the same port that the configuration app would be listening on. My
guess is that you'll either need to change the port that nginx is listening
to or the port that RaspiWiFi is listening to.

To change the the port that RaspiWiFi is listening to you either need to
edit line 22 in "Reset Device/static_files/rc.local.aphost.template" before
running the initial setup script or edit the "/etc/rc.local" file. You'll
need to change the "-p 80" and change the number 82 any other port number
not already used by another application on the system. In either case to
navigate to a web server on specific poor you need to form your url like
this:

10.0.0.1:84

This assumes that the web server you're trying to access is listening on
Port 84.

If you don't really need nginx for something else that you're doing you
could also just uninstall it with "sudo apt remove nginx".

Take a look into that and let me know what you figure out.

jasbur, 2018-02-21:
I forgot to mention that you can edit the "/etc/wpa_supplicant/wpa_supplicant.conf" directly to enter the information for your WiFi network manually if you just want to get it up and running quickly.

jasbur, 2018-02-27:
I just changed the default server port to 9191 in the latest version 1.4 release. That should clear up any of the issues with the conflicting with any other web server based application.

closed 2018-02-27 by jasbur