jasbur/RaspiWiFi #20: Trigger script start-up in a more friendly way

jasbur, 2018-03-01:
I'm currently starting all scripts (Both in Configuration and Client mode) by copying over the system's `/etc/rc.local` file. The existing `rc.local` is being overwritten completely in this process. 

That means that any other modifications, outside the scope of RaspiWiFi will be lost. So that RaspiWiFi can peacefully co-exist with other projects a more sane way to do this needs to be implimented. My current thought is maybe to append a line using the `@reboot` trigger in the /etc/crontab.

bottleworks, 2018-03-07:
I would use init.  
Or more simply do a copy, then restore rc.local instead of wipe and replace.

jasbur, 2018-03-23:
I tried triggering startup using systemd services, but it felt like over-engineering to me. I ended up triggering by using a `run-parts` line in the `/etc/crontab` instead.

closed 2018-03-23 by jasbur