jasbur/RaspiWiFi #21: Raspiiwi Setup found by device, but cannot connect to AP.

SidneyNiccolson, 2018-03-19:
Hi there,

Your project is very interesting. I installed the application on my Raspbian Stretch OS, but can't seem to connect to the

jasbur, 2018-03-19:
Based on your description my best guess would be that the DHCP server is failing to issue and IP address for some reason. Can you send me your syslog at /var/log/syslog from the Pi? That's where any DHCP errors would be logged and I might be able to figure something out.

SidneyNiccolson, 2018-03-19:
Thanks for your quick reply. The issue was indeed related to the DHCP server, namely the following messages "no wireless interfaces found" and "Not configured to listen to any interface". The log contains the following outstanding errors: 

> 
   118	Mar 19 16:04:37 raspberrypi dhcpd[409]: No subnet declaration for wlan0 (no IPv4 addresses).
   119	Mar 19 16:04:37 raspberrypi dhcpd[409]: ** Ignoring requests on wlan0.  If this is not what
   120	Mar 19 16:04:37 raspberrypi dhcpd[409]:    you want, please write a subnet declaration
   121	Mar 19 16:04:37 raspberrypi dhcpd[409]:    in your dhcpd.conf file for the network segment
   122	Mar 19 16:04:37 raspberrypi dhcpd[409]:    to which interface wlan0 is attached. **
   123	Mar 19 16:04:37 raspberrypi dhcpd[409]: 
   124	Mar 19 16:04:38 raspberrypi dhcpd[409]: 
   125	Mar 19 16:04:38 raspberrypi dhcpd[409]: Not configured to listen on any interfaces!
......
  136	Mar 19 16:04:40 raspberrypi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited status=1
   137	Mar 19 16:04:40 raspberrypi systemd[1]: isc-dhcp-server.service: Unit entered failed state.
   138	Mar 19 16:04:40 raspberrypi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
   139	Mar 19 16:05:08 raspberrypi systemd[1]: Starting dhcpcd on all interfaces...
   140	Mar 19 16:05:09 raspberrypi dhcpcd[310]: Not running dhcpcd because /etc/network/interfaces
   141	Mar 19 16:05:09 raspberrypi dhcpcd[310]: defines some interfaces that will use a
   142	Mar 19 16:05:09 raspberrypi dhcpcd[310]: DHCP client or static address
   143	Mar 19 16:05:09 raspberrypi systemd[1]: dhcpcd.service: Control process exited, code=exited status=6
   144	Mar 19 16:05:09 raspberrypi systemd[1]: Failed to start dhcpcd on all interfaces.
   145	Mar 19 16:05:09 raspberrypi systemd[1]: dhcpcd.service: Unit entered failed state.
   146	Mar 19 16:05:09 raspberrypi systemd[1]: dhcpcd.service: Failed with result 'exit-code'.

> 

I started the DHCP server manually (sudo service isc-dhcp-server start) and that made the AP connection work immediately. Afterwards the log showed this:

>    170	Mar 19 16:18:43 raspberrypi dhcpd[419]: No subnet declaration for usb0 (no IPv4 addresses).
   171	Mar 19 16:18:43 raspberrypi dhcpd[419]: ** Ignoring requests on usb0.  If this is not what
   172	Mar 19 16:18:43 raspberrypi dhcpd[419]:    you want, please write a subnet declaration
   173	Mar 19 16:18:43 raspberrypi dhcpd[419]:    in your dhcpd.conf file for the network segment
   174	Mar 19 16:18:43 raspberrypi dhcpd[419]:    to which interface usb0 is attached. **
   175	Mar 19 16:18:43 raspberrypi dhcpd[419]: 
   176	Mar 19 16:18:43 raspberrypi dhcpd[419]: 
   177	Mar 19 16:18:43 raspberrypi dhcpd[419]: Not configured to listen on any interfaces!
   178	Mar 19 16:18:43 raspberrypi dhcpd[419]: 
   179	Mar 19 16:18:43 raspberrypi dhcpd[419]: If you think you have received this message due to a bug rather
   180	Mar 19 16:18:43 raspberrypi dhcpd[419]: than a configuration issue please read the section on submitting
   181	Mar 19 16:18:43 raspberrypi dhcpd[419]: bugs on either our web page at www.isc.org or in the README file
   182	Mar 19 16:18:43 raspberrypi dhcpd[419]: before submitting a bug.  These pages explain the proper
   183	Mar 19 16:18:43 raspberrypi dhcpd[419]: process and the information we find helpful for debugging..
   184	Mar 19 16:18:43 raspberrypi dhcpd[419]: 
   185	Mar 19 16:18:43 raspberrypi dhcpd[419]: exiting.
   186	Mar 19 16:18:45 raspberrypi isc-dhcp-server[368]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ... failed!
   187	Mar 19 16:18:45 raspberrypi isc-dhcp-server[368]:  failed!
   188	Mar 19 16:18:45 raspberrypi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited status=1
   189	Mar 19 16:18:45 raspberrypi systemd[1]: isc-dhcp-server.service: Unit entered failed state.
   190	Mar 19 16:18:45 raspberrypi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
   191	Mar 19 16:18:49 raspberrypi dhcpcd-run-hooks[480]: wlan0: starting wpa_supplicant
   192	Mar 19 16:18:50 raspberrypi dhcpcd-run-hooks[488]: wlan0: failed to start wpa_supplicant
   193	Mar 19 16:18:50 raspberrypi dhcpcd-run-hooks[489]: wlan0: Successfully initialized wpa_supplicant#012Line 8: invalid cipher 'CCMP PSK'.#012Line 8: failed to parse pairwise 'CCMP PSK'.#012Line 11: failed to parse network block.#012Failed to read or parse configuration '/etc/wpa_supplicant/wpa_supplicant.conf'.
   194	Mar 19 16:18:51 raspberrypi dhcpcd[314]: wlan0: waiting for carrier
   195	Mar 19 16:18:51 raspberrypi dhcpcd[314]: wlan0: carrier acquired
   196	Mar 19 16:18:51 raspberrypi dhcpcd[314]: DUID 00:01:00:01:21:44:2c:65:b8:27:eb:8b:99:05
   197	Mar 19 16:18:51 raspberrypi dhcpcd[314]: wlan0: IAID eb:8b:99:05
   198	Mar 19 16:18:51 raspberrypi dhcpcd[314]: wlan0: adding address fe80::f986:10c3:9fdb:8e77
   199	Mar 19 16:18:51 raspberrypi dhcpcd[314]: wlan0: carrier lost
   200	Mar 19 16:18:51 raspberrypi dhcpcd[314]: wlan0: deleting address fe80::f986:10c3:9fdb:8e77

> 

I'm still trying to solve my network issue as after RaspiWifi reboot I cannot connect. I may have to use a fresh Stretch due to the configuration issues.

jasbur, 2018-03-19:
That's interesting, I anticipated some problems using the `rc.local` to trigger the software so I've been working on triggering it through `@reboot` in `/etc/cron` already. I'll probably be able to put the finishing touches on that later today. It's possible that everything is starting too soon or something.

Also, I think I'm going to make the same changes you made to that `temp_conf_file` since I'm pretty sure those two lines you removed aren't really necessary anyway.

For the record, which version of Raspberry Pi and Raspbian are you using? I've been testing my code changes on a Raspberry Pi 3 with Stretch all weekend and haven't had any issues at all.

SidneyNiccolson, 2018-03-19:
Indeed seeing your previous issue, Cron might do the job better. Maybe a short delay in cron helps. I hope to start testing again next week. My goal is to add more configurations to the view for mainly adding config files (json) for websocket client purposes to connect to a Django app. That should be doable right? I noticed some writing functions for wifi already in place.

I'm using the Zero W with stretch.

jasbur, 2018-03-20:
I was able to test a bit and it seems like it might be an issue with the drastically lower performance on the Zero W. If I manually delay the isc-dhcp-server service from starting by just 5 seconds everything works consistently.

For now, if you manually add `sleep 5` anywhere below the `#!/bin/bash` line in `/etc/init.d/isc-dhcp-server` it should fix the problem. I'm seeing some other discussion about this DHCP server starting in the wrong order too. I'll do a little research and try to come up with a better/more automatic fix.

SidneyNiccolson, 2018-03-23:
Thanks, will try it out! I'll keep tracking the project to see when the better fix out.

jasbur, 2018-03-23:
I've just released version 1.5 that has this fixed implemented. It seems to have been an issue with the slower CPU in the Pi Zero causing the DHCP service to start before the wireless interface was actually up and running. I just added a simple re-trigger for the DHCP service later in the boot process in case it failed.

closed 2018-03-23 by jasbur