jasbur/RaspiWiFi #22: isc-dhcp-server Service doesn't start correctly on Pi Zero W most of the time.

jasbur, 2018-03-20:
It seems to be starting out of order and trying to launch before the network interfaces are ready. A manual work-around is to add `sleep 5` just below the `#!/bin/bash` in `/etc/init/networking.conf`

This should be automated somehow during the execution of the `initial-setup.py` script.

pyrosmiley, 2018-03-20:
Out of curiosity I tried to do the manual work-around on a pi0w, but looking at my ```/etc/init/networking.conf```, I can't actually find where the shebang is? It has pre-start scripts and post-stop scripts but no ```#!/bin/bash```, and I don't understand it well enough to know where to look.

jasbur, 2018-03-20:
Oopsie, I gave the wrong path in the description. You need to modify `/etc/init.d/isc-dhcp-server` as noted above.

pyrosmiley, 2018-03-20:
AHHHH okay. That makes sense!

jasbur, 2018-03-20:
Just updated the original issue accordingly too.

jasbur, 2018-03-23:
This is a duplicate of Bug #21 so I'll just paste the same comment about it here. It should be fixed in the 1.5 release I just posted.

"I've just released version 1.5 that has this fixed implemented. It seems to have been an issue with the slower CPU in the Pi Zero causing the DHCP service to start before the wireless interface was actually up and running. I just added a simple re-trigger for the DHCP service later in the boot process in case it failed."

closed 2018-03-23 by jasbur