jasbur/RaspiWiFi #24: Application misconfigures a WPA2 access point as WPA

Anomalyzero, 2018-03-29:
Hello, first off much thanks for putting this project together. 

I've been trying to get this working for my home wifi, but just couldn't get it going right. I tried it at friend's houses though, and discovered it worked there, so I set about figuring out what was wrong with mine.

I eventually discovered that it had configured `wpa_supplicant.conf` file as WPA instead of WPA2. By removing the lines below, the Pi would connect (after a restart).
```
pairwise=CCMP PSK
group=CCMP TKIP
```

I'm stilling working on why exactly it did this. Unfortunately my knowledge of Linux networking is rather limited, so I haven't been able to ascertain why yet. I wanted to get this bug report in while I investigate it further. I'll report back when I know more.

jasbur, 2018-03-29:
WPA supports TKIP and AES variants of the encryption protocol. However some
routers only seem to allow AES. You've discovered the right fit for the
problem though! Thanks for the heads up. I'm planning a 1.5.1 release soon
and I'll incorporate your fix.

Anomalyzero, 2018-03-29:
So, I checked my router's output for iwlist scan. It looks like my particular network is outputting both of the strings `main.rb` looks for when setting `encryption_type`. See below:

```
          Cell 02 - Address: <macaddr>
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=68/70  Signal level=-42 dBm
                    Encryption key:on
                    ESSID:"MyWifiAP"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 80ms ago
                    IE: Unknown: 001554502D4C494E4B5F322E3447487A5F454545314446
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030106
                    IE: Unknown: 2A0100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1AAD0103FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 331AAD0103FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1606051500000000000000000000000000000000000000
                    IE: Unknown: 341606051500000000000000000000000000000000000000
                    IE: WPA Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: DD180050F2020101850003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD9B0050F204104A0001101044000102103B0001031047001000010203040506070809C04A00EEE1E01021000754502D4C494E4B1023000A544C2D5744523433303010240003312E3010420003312E301054000800060050F20400011011001A576972656C65737320526F7574657220544C2D57445234333030100800020086103C000101104900140024E26002000101600000020001600100020001
```

Anomalyzero, 2018-03-29:
It's something with my particular router... I have the option to select `Automatic` for encryption scheme instead of WPA/WPA2. When Automatic is selected, both of those strings appear in my output for `iwlist scan`

jasbur, 2018-03-29:
Great, thanks for the details. Now to try to find a delicate enough fix that it doesn't break everything else :)

jasbur, 2018-04-29:
I ended up simplifying the `wpa_supplicant.conf` that was being generated during the configuration. It turns out less is more here. If we just add the SSID and Network Key to the `wpa_supplicant.conf` the wpa_supplicant service does all of the difficult negotaion about key type.

Now the RaspiWiFi basically just detects if the access point is encrypted or not and generates the `wpa_supplicant.conf` accordingly.

Currently the fix is in the [v1.5.1_alpha branch](https://github.com/jasbur/RaspiWiFi/tree/v1.5.1_alpha) | [here](https://github.com/jasbur/RaspiWiFi/blob/b2fbc31f9039c974025c0f1024c576899374586c/Configuration%20App/app/models/main.rb#L65). I'll be merging this branch into the master in the near future for release.

closed 2018-04-29 by jasbur

jasbur, 2018-05-08:
Just a heads up, I just release [v1.6](https://github.com/jasbur/RaspiWiFi/releases/tag/v1.6) which should fix your issue. Let me know if you have any problems with it.