jasbur/RaspiWiFi #25: Error when installing on Ubuntu Mate

Jordandarabos, 2018-04-13:
`Bundle complete! 11 Gemfile dependencies, 53 gems now installed.
Use `bundle show [gemname]` to see where a bundled gem is installed.

cp: cannot stat 'Reset Device/static_files/apclient_bootstrapper.template': No such file or directory
cp: cannot stat 'Reset Device/static_files/aphost_bootstrapper.template': No such file or directory
cp: cannot stat 'Reset Device/reset.py.template': No such file or directory
cp: cannot stat 'Reset Device/manual_reset.py.template': No such file or directory
Traceback (most recent call last):
  File "/home/zabatt/RaspiWiFi-master/initial_setup.py", line 64, in <module>
    update_config_paths()
  File "/home/zabatt/RaspiWiFi-master/initial_setup.py", line 29, in update_config_paths
    for line in file:
  File "/usr/lib/python3.5/fileinput.py", line 248, in __next__
    line = self._readline()
  File "/usr/lib/python3.5/fileinput.py", line 335, in _readline
    os.rename(self._filename, self._backupfilename)
FileNotFoundError: [Errno 2] No such file or directory: 'Reset Device/static_files/aphost_bootstrapper' -> 'Reset Device/static_files/aphost_bootstrapper.bak'
`

Any suggestions?

jasbur, 2018-04-13:
I haven't tested at all on a non-Raspbian OS yet. I should probably have a little time this weekend to try it out and maybe come up with a fix.

jasbur, 2018-04-14:
I just pushed a branch called "ubuntu_fix". I don't have the time to test it just yet, but I think it might work. You can download it from here:

https://github.com/jasbur/RaspiWiFi/archive/ubuntu_fix.zip

Give it a try if you have the time and let me know if it works for you.

jasbur, 2018-04-14:
You might want to try with a fresh image if possible.

jasbur, 2018-04-16:
I just tried it and it doesn't work quite yet. I'll let you know when i figure something out. In the meantime you could use Raspbian and the Main branch if it's an option for you.

Jordandarabos, 2018-04-16:
Ok, I haven’t been able to get to it this weekend. Thanks for the update.
Any idea what’s breaking it? I’d be happy to help debug it. I’ll give it a
shot first thing in the morning.

Jordandarabos, 2018-04-16:
> Yeah, Ubuntu appearently doesn't use the
> /etc/wpa_supplicant/wpa_supplicant.conf like Raspbian does. I'm trying to
> use the Network Manager from the command line using
>
> nmcli device wifi connect [SSID] password [PASS]
>
> on line 106 in RaspiWiFi/Configuration App/app/models/main.rb after
> detecting Ubuntu in the output from lsb_release -a
>
> The line seems to be triggering correctly, but not getting the job done
> while running the command manually from the terminal works fine.

Hmm ok I’ll try it in the morning and let you know how that goes. Thanks!

Jordandarabos, 2018-04-17:
Good morning,

So I finally got around to trying it and I am getting the same issue.I
noticed that there is a test/development environment in the code. How do I
initiate that so I can see the debugging? Otherwise the code moves so fast
on the command line I can't see anything lol. Thank you again for your help!

jasbur, 2018-04-17:
Open a terminal on the pie and issue this command to kill the already running rails server:

`sudo killall ruby`

Then navigate to the `Configuration App` folder on the RaspiWiFi directory and issue:

`sudo rails s`

To start the server in development mode. All the booking information will appear in the terminal window and you'll also get more verbose output through the web interface.

jasbur, 2018-04-17:
I forgot to mention, the development server will defaults to Port 300ⁿ port 9191. If you issue:

`sudo rails s -b 0.0.0.0 -p 9191`

It will start on Port 9191 like a dozen the normal operation. The `-b` switch will make sure you can access it from other devices on your network as well.

Jordandarabos, 2018-04-17:
=> Booting WEBrick
=> Rails 4.2.0 application starting in development on http://0.0.0.0:9191
=> Run `rails server -h` for more startup options
=> Ctrl-C to shutdown server
[2018-04-17 10:30:17] INFO  WEBrick 1.3.1
[2018-04-17 10:30:17] INFO  ruby 2.3.1 (2016-04-26) [arm-linux-gnueabihf]
[2018-04-17 10:30:17] INFO  WEBrick::HTTPServer#start: pid=1526 port=9191

This is all it does I don't see it broadcasting at all on other devices.
Thanks

jasbur, 2018-04-17:
Actually, it looks like the server started OK. You just need to navigate to it from a web browser now. On the Pi itself you can get there by opening the web browser and entering

`localhost:9191`

At this point it may not be broadcasting the hotspot. Those are actually two different functions. For debugging purposes you'd want to be on the Pi anyway.

Jordandarabos, 2018-04-17:
This is what I’m getting. Sorry I’m not too familiar with Ruby on Rails.

Jordandarabos, 2018-04-17:
Ok update....

I just tried to install it on the latest version of Raspian Stretch
(desktop version) and I am getting the same results as with the Ubuntu Mate
installation. It appears to only work on raspian lite. Going to try it on
Ubuntu snappy core.

jasbur, 2018-06-11:
I'm going to close this for now. I'll be focussing on Raspbian for the foreseeable future. At some point I'll certainly revisit compatibility with other distributions such a MATE though.

closed 2018-06-11 by jasbur