jasbur/RaspiWiFi #26: dhcpcd.service fails to start in AP Host mode

zachdolan16, 2018-04-23:
On a fresh install of (at least) these two versions of Raspbian **Stretch** [2017-11-29](http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-12-01/) and [2018-04-18](http://downloads.raspberrypi.org/raspbian/images/raspbian-2018-04-19/) the dhcpcd service fails to start. Meaning the RPI doesn't turn into ap host mode.

Looking at the status of the dhcpcd service:

[image]

After deleting /etc/network/interfaces and rebooting the RPI, the dhcpcd service starts and the RPI turns into ap host mode. But deleting the interfaces file removes the static IP which is very useful for ap host mode. I'm able to access the WiFi setup portal page on http://169.254.22.70:9191 . So, a static IP will need to be setup somewhere else?

In [main.rb](https://github.com/jasbur/RaspiWiFi/blob/73bd5f2af529b64cd06ecca773e0c0e26667a648/Configuration%20App/app/models/main.rb#L111) you have checking for Raspbian Stretch, to delete the interfaces file. So, checking for Raspbian Stretch will need to be added into the initial_setup, reset, and manual_reset?

jasbur, 2018-04-23:
Good point about the reset functions needing a version check as well. But didn't even occur to me until you just posted that. I should get a chance on Wednesday, most likely, to take a look at this in the other issues that have appeared. I'll post here when I figure out a good way to make all this work. Thanks for the feedback.

jasbur, 2018-04-25:
Ok, so I just re-tested v1.5 with Raspbian Jessie Lite and Desktop images and had no issues at all. For both I downloaded the current image from https://www.raspberrypi.org/downloads/raspbian/ , downloaded the ZIP of the current `master` branch (which is the same as the v1.5 release), extracted it and ran `sudo python3 initial_setup.py` from the extracted directory.

On reboot the whole configuration process worked fine and rebooted itself into a working connection. I even tried the reset via GPIO 18 and it worked without issue. Can you think of anything that might be unique about your install? Any changes that you made prior to running the setup?

Also, I looked into the `interfaces` file being deleted on reset; it occured to me that it's only relevant when switching the device into `client` mode so the OS check is only necessary in the `main.rb` because every other function will set it to `Host` mode.

zachdolan16, 2018-04-25:
> I just re-tested v1.5 with Raspbian Jessie Lite and Desktop images

I assume you meant Raspbian Stretch? Since it is the latest from https://www.raspberrypi.org/downloads/raspbian/  

I just re tested this again. and getting the same result of the dhcpcp.service not starting.

My process:

- Download 2018-04-18-raspbian-stretch.img which is the latest.
- Format SD card
- Write img to SD card
- Plug in RPI
- Enter WiFi Details
- Download the master branch, extract to /home/pi/Desktop/RaspiWiFi/
- Ran initial setup (sudo python3 initial_setup.py)
- Reboot RPI as per initial setup

And like before the dhcpcd.service fails to start exactly like the picture in my initial post.

I did the reset via GPIO 18, and the dhcpcd.service still fails to start like above.

I put the SD card into a different RPI (Raspberry Pi 3 Model B V 1.2 on the board itself), and the dhcpcd.service fails to start.

You said you used Raspbian Jessie, which I assume is a typo. But if you used Jessie instead of Stretch, that would be the reason why you aren't having the same problems that I am.

jasbur, 2018-04-25:
Yeah, Jessie was a typo I meant to type Stretch.

I know previous to version 1.5 the DHCP service not starting correctly was an issue on the lower powered Pi devices like the Zero. I added a workaround for the problem in 1.5 and a test successfully on my zero at least. 

Have you tried using the pre-formatted image on the release page for 1.5 yet?

jasbur, 2018-04-27:
Great to hear! Thanks for all of the follow up and testing. I'll close this issue for now, but I'll keep an eye out for the potential cause of that error.

closed 2018-04-27 by jasbur