jasbur/RaspiWiFi #29: Update reset_all function on Web Page

zachdolan16, 2018-04-27:
The [self.reset_all in main.rb](https://github.com/jasbur/RaspiWiFi/blob/73bd5f2af529b64cd06ecca773e0c0e26667a648/Configuration%20App/app/models/main.rb#L121) hasn't been updated from the old /etc/rc.local functionality. 

Realistically this function can just call the manual_reset.py file.

jasbur, 2018-04-27:
Actually this is deprecated code left over from when I used to have a "Reset" button on the web interface. It was really there more for debugging purposes in the early days of the project and I took out the button interface long ago.

Thanks for the heads-up though. I'll be removing that code block before the next release now that you've reminded me that it's still there.

jasbur, 2018-04-29:
I've just removed all of the deprecated code relating to `reset_all` in the [v1.5.1_alpha branch](https://github.com/jasbur/RaspiWiFi/tree/v1.5.1_alpha). Thanks for the heads up.

closed 2018-04-29 by jasbur