jasbur/RaspiWiFi #30: Error when SSID ends in a '+'

zachdolan16, 2018-05-04:
I was testing out RaspiWiFi with a mobile hotspot that was created from my phone. The default SSID for my phone was "Zachs Galaxy S8+". The error is that the '+' sign gets ignored since the save_credentials controller thinks the '+' sign is seperating the SSID and the encryption type.

In my situation, the value of the <option> in the dropdown list was something like "Zachs Galaxy S8++open". So, when the server side function processes that value, it determines that the SSID is "Zachs Galaxy S8", when in actuality it's missing the '+' sign. 

https://github.com/jasbur/RaspiWiFi/blob/791557187c09683f57f87ec5eb6b89be42a9ef72/Configuration%20App/app/controllers/main_controller.rb#L9-L10

Any delimiter that you chose to stick between the SSID & the encryption type in the <option> value would cause interference, so a different solution for handling SSID & encryption type would need to be implemented. Something like [this solution would work](https://stackoverflow.com/a/14579469/3017977). I don't know how Ruby so I can't implement the server side handling of this, but something like this would work for the HTML side of things with the answer from that StackOverflow answer.

`<option value='<%=ap[:ssid]%>' encryption_type='<%=ap[:encryption_type]%>'><%=ap[:ssid]%></option>`

jasbur, 2018-05-05:
I was chugging through a solution to this problem when it occurred to me that I already made a change in way the `/etc/wpa_supplicant/wpa_supplicant.conf` that makes determining the encryption type unnecessary. Now the only thing that's important is to determine whether or not encryption exists at all.

Pulling out the code determining specific encryption type should eliminate the need to use a delimiter at all.

jasbur, 2018-05-08:
Just a heads up, I just release [v1.6](https://github.com/jasbur/RaspiWiFi/releases/tag/v1.6) which should fix your issue. Let me know if you have any problems with it.