jasbur/RaspiWiFi #32: loose ethernet Connection

rajkiranjeet, 2018-05-08:
Hi
i have installed RaspiWiFi install ok and access pint created.problem is i dont want to connect wifi now i want to choose ethernet till i require ,but in this case my existing DHCP ip loose .
is it poosbil;e keep DHCP on till user connect to AP.
please let us know.
thanks

jasbur, 2018-05-08:
Your right, currently RaspiWiFi does cause the Ethernet port not to work properly. I don't have a work-around to offer you, but I'll add it to the list of things to fix for the next release.

I'll reply here too if I come across and easy fix you could use in the meantime.

rajkiranjeet, 2018-05-09:
Hi
one more thing i noticed after setting wifi it takes 2 ips one for Ethernet and one for wifi.
so it should give first priority to Ethernet ,id not available then go to wifi.

jasbur, 2018-05-17:
I've just finished converting all DHCP functions from the `isc-dhcp-server` to `dnsmasq` in the [dnsmasq_conversion branch](https://github.com/jasbur/RaspiWiFi/tree/dnsmasq_conversion). This should resolve all issues that you were having with the Ethernet port, both in AP Host and AP Client mode.

I'll be merging the changes into the main branch as soon as I finish the other changes for the next release.

closed 2018-05-17 by jasbur

rajkiranjeet, 2018-05-17:
thanks ,
is it available now or should we wait for next release ?

jasbur, 2018-05-17:
You could download an archive with all of the changes here:

https://github.com/jasbur/RaspiWiFi/archive/dnsmasq_conversion.zip

You'll just need to run

`sudo python3 initial_setup.py`

from the extracted directory. You'll need an internet connection during the setup. If you prefer a ready-to-go image you'll have to wait until the next release.

jasbur, 2018-06-11:
I just wanted to let you know that the Ethernet ports will now remain active at all times in from current release [v1.7](https://github.com/jasbur/RaspiWiFi/releases/tag/v1.7) going forward. Thanks for the bug report!

esbennn, 2018-10-25:
Hi!
This tool is great and does it's job well. However, though this problem has been solved, a similar problem seems to exist concerning 3/4G usb dongles. 
After installing this, the dongle which previously worked, does not. It will connect to a gsm network, but attempting a ping returns a name resolution error.

Any chance you have an idea to fix that?

esbennn, 2018-10-26:
Thanks for the quick reply @jasbur 
Pinging www.google.dk gives `ping: www.google.dk: Temporary failure in name resolution`
Pingin 8.8.8.8 results in `From 169.254.62.145 icmp_seq=XX Destination Host Unreachable`
Does that tell you anything?

jasbur, 2018-11-19:
Okay, it's clearly not just a problem with DNS resolution then. I guess I don't know enough about configuring Raspbian to work with that kind of hardware.

Can you reply with a link or anything that might give me a little bit of detail?