jasbur/RaspiWiFi #33: Connect to a hidden network

tabris2015, 2018-05-17:
Hi! I would like to know how can I connect to a hidden wifi network or how can i change the code to help me to do so. Thanks!

jasbur, 2018-05-17:
Sure, that shouldn't be too difficult to add. I'm in the process of re-tooling the DHCP server at the moment for `v1.7`, but I'll add in manual SSID entry for that release right after I'm done.

jasbur, 2018-06-13:
I just added a link below the SSID selection list that will allow you to enter an SSID manually. It's currently avaiable in the [flask_config_conversion](https://github.com/jasbur/RaspiWiFi/tree/flask_config_conversion) branch and will be pushed to the `master` branch later today. It will also be available in the upcoming `v1.8` release which will be released soon.

Thanks for the suggestion.

closed 2018-06-13 by jasbur