jasbur/RaspiWiFi #34: Multiple Wifi networks.

charely6, 2018-05-25:
I haven't tried it out yet but this is exactly the program I am looking for, I was wondering if you have looked into making it capable of learning multiple wifi networks, so if it can't find a wifi network it knows within a certain time it switches to config mode to learn a new one or remove old ones if needed.

jasbur, 2018-05-26:
I'm going to be looking into this soon. Thanks for the suggestion.

jasbur, 2018-06-06:
I'll be issuing a new release later today that will have this functionality. During the setup you'll be able to decide whether or not you'd like to enable the auto-reconfiguration function and how many seconds you'd like it to wait before triggering.

You'll also be able to manually edit those values in the new `/etc/raspiwifi/raspiwifi.conf` file.

closed 2018-06-06 by jasbur

jasbur, 2018-06-06:
Ok, I just added the new release. Let me know if you have any issues.