jasbur/RaspiWiFi #35: Wifi Country

rboorman, 2018-05-26:
Thank you for this tool - its awesome.   I had typed in my PSK wrong so i in Pi world - bricked my device.  Had to actually work and put a console cable on. lol ;)  - Of course you handled this with a reset button too - (do you have some instructions on how to wire the button up?)

During this time of resetting - i noticed an warning on login that Wifi was disabled due to wifi country was not selected.

I added this to mine but maybe cool to get a country selection during install of the tool.  Right now I edited ./app/models/main.rb

temp_conf_file.puts 'country=US'

Added that to my setup to prevent the message (it may be nothing to worry about at all - i know it only shows when we are resetting configuration and wpa_supplicant gets deleted.

Thank you

jasbur, 2018-05-26:
[This](https://projects.raspberrypi.org/en/projects/physical-computing/7) diagram illustrates it pretty well. Basically, all you need is to bridge on of the 3.3v pins on the Pi with GPIO18 to count as a press. No need for fancy buttons, I just hold 2 wires together when I'm testing :)

Keep that 3.3v on GPIO18 for 10 seconds and everything will reset. 

I'm actually working on the code at this moment trying to add in more configuration, etc. So, I'll take a look at the Country Specification as soon as I get a chance. I haven't had the issue myself but there seem to be a lot of small quirks with different WiFi configs out there.

Also, I'm working on an "Auto Config" trigger which will re-trigger the AP Host mode when a problem with the connectivity is detected. That would help in the type of situation you described above.

rboorman, 2018-05-26:
Thank you! - Im actually going to use a button for my simulation of things.   Looks like we are on the same page for sure on how to take this to the next level.  Here is a crazy thought - I want to have several pi's and potentially other devices where the main AP can have its button pressed 2 or more consecutive times and it will enter into BLE to BLE join - so it will transfer configuration of wifi over to other pi's.  The reason for BLE is just to exercise location based/nearby requirements - we can set RSSI really low to where the pies have to touch to transfer.  Anyway - you can kind of see where my heads at.  - Great work!

jasbur, 2018-05-26:
That's a cool idea. I might look into that when I get his next release done. I'm was thinking about maybe using BLE and a phone app down the road as a second method of configuration anyway. Sound like fun!

zachdolan16, 2018-06-05:
> re-trigger the AP Host mode when a problem with the connectivity is detected.

Does this include a want to detect an incorrect password (PSK) ? I created a script which does this, and turns the device back into AP Host Mode. I'm thinking that you might be able to create a better way to detect it though, maybe even through the web page? (try to connect to the wpa_supplicant before attempting to turn device into AP client mode??)

In my use case for this project, is that I would disable the resetting if it detects connectivity issues. But still have the password checking somewhere, whether it be on the web page itself, or in another script somewhere.

Here is my script which handles resetting to AP Host Mode on incorrect password detected:

```bash
#!/bin/bash
#2018-04-27

#Checks the credentials in the wpa_supplicant.conf file. If they are incorrect, it resets the device back into
#AP Host Mode, allowing the user to re configure a WiFi network.

connection_result=false
#Since this script is ran on @reboot from the crontab, you need to sleep a couple seconds for the wlan interface to
#be configured or something along those lines.
sleep 10

function checkWiFi {
	#Start process which will output if authenticated properly.
	#the command is actually at the end of the while loop, (the "done" command)
	logger "RaspiWiFi: Starting new wpa_supplicant process to check WiFi credentials" -s
    while read line
	do
		#echo "$line"
		#https://superuser.com/a/903893
		#Run the wpa_supplicant to attempt to connect, if it wrong password/PSK, '4-Way Handshake failed' will appear
		#If password/PSK is correct, 'CTRL-EVENT-CONNECTED' will appear
		#Read line by line because the cmd output isn't instant.
		logger "RaspiWiFi: $line" -s
		case "$line" in
			*'4-Way Handshake failed'*)
				logger "RaspiWiFi: Incorrect password/PSK provided" -s
				connect_result=false
				return
		;;
			*'CTRL-EVENT-CONNECTED'*)
				logger "RaspiWiFi: Correct PSK/password provided" -s
				connect_result=true
				return
		;;
		esac
	#https://raspberrypi.stackexchange.com/a/61137
	#This checks the WiFi credentials.
	done < <(sudo wpa_supplicant -D wext -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlan0 -d)
	return
}

logger "RaspiWiFi: Running CheckWiFiCredentials" -s
if [ -f /etc/wpa_supplicant/wpa_supplicant.conf ];
then
	#Kill the existing process because you cannot start multiple wpa_supplicant processes at once.
	logger "RaspiWiFi: Killing the existing wpa_supplicant process" -s
	sudo pkill -f wpa_supplicant
	sleep 1
	logger "RaspiWiFi: Checking credentials for the given WiFi" -s

	#this function runs the wpa_cli to check the password, and sets the connection_result variable.
	checkWiFi

	logger "RaspiWiFi: Killing wpa_supplicant process that was used for checking credentials" -s
	sudo pkill -f wpa_supplicant

	logger "RaspiWiFi: Connection Result: $connect_result" -s
	if [ "$connect_result" = true ];
	then
		logger "RaspiWiFi: Credentials were correct, and connected successfully. Restarting wpa_supplicant process" -s
		wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0 -Dnl80211,wext
	else
		logger "RaspiWiFi: Incorrect password/PSK provided. Turning unit back into AP Host Mode, running manual_reset.py" -s
		sudo python3 /home/pi/Desktop/RaspiWiFi/Reset\ Device/manual_reset.py
	fi
else
	logger "RaspiWiFi: Unit in AP Host Mode (wpa_supplicant.conf is missing). Not doing anything." -s
fi
```

jasbur, 2018-06-06:
I'm going to issue a release today that will have an auto monitoring (enabled optionally) function to monitor whether the connection is valid or not and reset to Host Mode after a certain period of time (configurable). For now, this will work as a fallback if the wrong PSK was entered. Those values can be configured during the setup process now or by editing the new `/etc/raspiwifi/raspiwifi.conf`.

I'm going to try to use your code as a guide to implement a smarter way of checking the PSK for the next release though. Ideally there should be a check immediately after clicking the `Connect` button on the configuration page with a warning and a chance to re-enter the key before rebooting the Pi blindly.

I'll keep you updated, but in the meantime watch for the release later today.

jasbur, 2018-06-06:
Ok, I just added the new release. Let me know if you have any issues.

jasbur, 2018-06-10:
I'm going to close this issue since it seems like the new auto-configuration routine would prevent someone being locked out if they enter incorrect information. I'm still going to be looking at immediate credential verification and the BLE suggestion you made though.

closed 2018-06-10 by jasbur