jasbur/RaspiWiFi #40: Not working for me

hbattat, 2018-06-19:
I tried using the script to install, and I was not able to see an AP. Then I started fresh with your IMG file, still no luck, am I missing something?
This is on a RPI B+ and  with a Realtek 8188EUS Wifi dongle

Thanks for your help, this is a great solution (when it works for me :) )

jasbur, 2018-06-19:
I've only tested the software with Raspberry Pi 3, Raspberry Pi zero W, and a Raspberry Pi B+ using an Edimax Wi-Fi adapter. My guess is that the chipset your Realtek adapter is using may not be compatible with hostapd, which is used to generate the access point.

It's possible that the current configuration for hostapd is a little too specific. Try deleting this line:

`driver=nl80211`

From `/etc/hostapd/hostapd.conf` and `/usr/lib/raspiwifi/reset_device/static_files/hostapd.conf` After that reboot the Pi back into configuration mode and see if it works.

jasbur, 2018-06-25:
Did you end up trying this fix out? Did it work for you?

hbattat, 2018-06-25:
It was my USB wifi dongle, It does seem like it is supported by hostapd

closed 2018-06-25 by hbattat