jasbur/RaspiWiFi #41: uninstalling this tool

tabris2015, 2018-06-24:
I am using a raspberry pi 3 with ubuntu mate installed and it seems like this tool doesn't work because there is no AP created, so I was wondering if there is a way of make a clean uninstall for trying to configure my wifi manually. Thanks!

aslafy-z, 2018-06-25:
I'm currently trying to build a `deb` package from this tool to be able to install and uninstall it in a clean way. I'm hitting some issues when pushing configuration files from other packages (dhcpcd.conf, hostapd.conf, dnsmasq.conf and wpa-supplicant.conf). I'm looking for a way of switching config files for these daemons without replacing core config files (eg: keep them all in /usr/lib/raspiwifi). If anyone has a trick to handle this it would be great.

jasbur, 2018-07-29:
I've just released version 1.8 and I'll be looking at implementing an uninstaller next.

jasbur, 2019-01-20:
I just added an Uninstaller to the project. If you update your repository or re-clone it you can run the Uninstaller like this:

`sudo python3 libs/uninstall.py`

Install from now on can always access it from `/usr/lib/raspiwifi/uninstall.py` as well.

closed 2019-01-20 by jasbur