jasbur/RaspiWiFi #43: Stops working after sudo pip3 install flask

nikivanov, 2018-06-27:
Steps to reproduce:
1. Use the latest RaspiWiFi image
2. Connect to your Wifi using RaspiWiFi
3. sudo apt-get install python3-pip
4. sudo pip3 install flask
5. sudo rm /etc/wpa_supplicant/wpa_supplicant.conf 
6. Reboot

After deleting wpa_supplicant.conf and rebooting, I don't see the RaspiWiFi access point anymore.

I have a python script that runs in sudo on startup, so it's easier for me to sudo pip3 install flask, but as a workaround, I'll try to use virtualenv to isolate the python3 flask.

jasbur, 2018-07-02:
Hmm, removing the wpa_supplicant shouldn't have any impact on the hot spot being broadcast. Hostapd (which creates the broadcast) is triggered from a line in the system crontab. Is there any chance when you're triggering your script that it's overwriting that?

also, as an interesting side note I've rewritten the web interface to use Flask in the upcoming release v1.8 . That would mean that Flask would already be installed for you. You could test it out by cloning the master branch for now.

jasbur, 2018-07-29:
I've just release version 1.8 which has Flask installed by default, so that should probably fix that issue for you.

Let me know if it works properly for you now.

jasbur, 2018-08-22:
I haven't heard back, so I'm going to close this issue.

closed 2018-08-22 by jasbur