jasbur/RaspiWiFi #44: Change ip address to domain

andrewdhastings, 2018-07-15:
Is there a way to change the ip:port address to a "domain"?

For instance, on ASUS routers, you can type in router.asus.com in a browser and it will point to the setup page. Just makes it easier to remember if it dosen't auto start then a long ip:port string.

jasbur, 2018-07-16:
It kind of is working already in the [captive_portal](https://github.com/jasbur/RaspiWiFi/tree/captive_portal) branch. It should resolve any domain entered to 10.0.0.1:9191 so, you should be able to enter something like `raspiwifisetup.com` and it should redirect.

I'm trying to get it to automatically open the configuration page though like what happens when you go to Starbucks or most hotels. That would be cool.

andrewdhastings, 2018-07-16:
Oh, cool I will have to try that branch. And yes that would be awesome if it could auto connect when a browser is launched. I was just testing the pre-configured image and it seemed to work great!

I have also been testing this option: https://github.com/resin-io/resin-wifi-connect but it didn't auto run when the pi was booted. Not sure if I did something wrong on the config, but had ssh in and run a command to execute the wifi config. Kinda defeats the point. lol. Also, they don't have a way to clear the ssid/passwords and no reset switch, so I really like yours! 

Yours also seems much simpler. 

Another thought, when I selected the SSID and the password and hit connect, it immediately gave me a page cannot load error. The resin version was able to keep the page alive and tell you that if the connection fails it will reopen the page, that would be a nice additional feature to add to this one.

jasbur, 2018-07-29:
I've just released version 1.8 with your recommendation included. You can now go to http://10.0.0.1 or http://raspiwifisetup.com

I also added an option to change the port for people that have conflicting servers running so there's no need to specify the port anymore (since it's now port 80, the default web port).

Next, I'll look into checking the connection status before rebooting.

closed 2018-07-29 by jasbur