jasbur/RaspiWiFi #46: After choosing Wifi Network, raspberry doesnt connect

chalamministries, 2018-08-02:
I've chosen my wireless network and it reboots but it never connects to the actual network. I've tried on different SSID's.

chalamministries, 2018-08-02:
It never tries to get a new IP. It still says:

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.1  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::ba27:ebff:febc:6a89  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:bc:6a:89  txqueuelen 1000  (Ethernet)
        RX packets 388  bytes 37411 (36.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 120  bytes 16484 (16.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mulyanaa, 2018-08-09:
hey @chalamministries. maybe you reset the device at  Configuration Mode, so /etc/dhcpcd.conf.original file will replaced by /etc/dhcpcd.conf file and your raspberry will read static IP

cchamikara, 2018-11-26:
I have the same issue