jasbur/RaspiWiFi #47: .IMG?

testubule, 2018-08-04:
where do I find the .IMG? thanks,

closed 2018-08-04 by testubule

aslafy-z, 2018-08-04:
Let's go to the Release tab 😘

testubule, 2018-08-04:
Thanks, first time on github. excited to try this out

aslafy-z, 2018-08-04:
Enjoy, it works pretty well

testubule, 2018-08-05:
This works like a dream with the IMG ; really well done & couldn't be simpler. Initially started with latest Raspbian Stretch image and the script install approach, but had same issue as some others where flask installation step didnt complete (& wpa_supplicant.conf wiped out).
Will try that again sometime in future.
Thanks, great product.

testubule, 2018-08-09:
I will roll back and redo the script install this weekend to capture logs. I will try the master branch so youll have exactly what I did before. Thanks for your help

testubule, 2018-08-11:
Fresh flash of "2018-06-27-raspbian-stretch", install of wpa_supplicant.conf to my wifi and then RaspiWiFi script install. No errors during script install (removed "clear" syntax from install script to observe the entire output during installation to confirm). Reboot completes; AP broadcasts; can connect to AP;
Thereafter cannot 10.0.0.1. Any recommendations on which logs would indicate whether restarting webserver was successful? thanks