jasbur/RaspiWiFi #48: Not connecting to web server

ngservices, 2018-09-05:
Hello,

I'm using a Raspberry pi 3
Latest Raspian stretch 2018/06/27 image
Fresh install + updates
Installed RaspiWiFi.

I get to the configuration page OK. I enter the relevant details.
1. Typed in custom SSID
2. selected auto recover
3. Left default port number and also tried custom port number. 
4. selected 200 s
5. Said no to SSL security

It installs the relevant files. Asks me to reboot the RPI3.  I do the reboot.
I then detect the AP ssid on my list of wifi devices on another computer. 
I connect to the Wifi hotspot as per my custom name. 
I try to use the 10.0.0.1:1234 and also with default port i just typed in the IP nothing happens. it says unable to connect. 
I have done a couple of fresh formats and restarts.  I havent tried your image yet.

Any idea what can be going on ? Would love some help.

Thanks Dan

ngservices, 2018-09-05:
Hello Jasbur. i got your image to work.
I think my confusion is the very start of your tutorial.
I run the script it asks me to enter the details before it downloads the files. I then copy the files to /usr/lib/raspiwifi/ reboot the pi? the pi will then detect these files and take me through the setup process again ?

 _**_SCRIPT-BASED INSTALLATION INSTRUCTIONS:_

== Navigate to the directory where you downloaded or cloned RaspiWiFi

== Run:

sudo python3 initial_setup.py

== This script will install all necessary prerequisites, copy the project files
to /usr/lib/raspiwifi/, copy configuration
files for AP Host (configuration) mode, and reboot. When it finishes booting it
should present itself in "Configuration Mode" as a WiFi access point with the
name "RaspiWiFi[xxxx] Setup".

== The original RaspiWiFi directory that you ran the Initial Setup is no longer
needed after installation and can be safely deleted. All necessary files are
copied to /usr/lib/raspiwifi/ on setup._**

jasbur, 2018-09-05:
Ah, I think I see to problem when the README says:

"This script will install all necessary prerequisites, copy the project files
to /usr/lib/raspiwifi/, copy configuration
files for AP Host (configuration) mode, and reboot."

I meant that the script will do all that for you then reboot, not that you have to do it. I'll try to update the wording a bit to be more clear. Thanks for the heads up.

jasbur, 2018-09-05:
I just pushed a slight wording change to the `Readme.txt` to be a bit more clear.

SmithtekWA, 2018-09-06:
Hi Jason.

I used the image you provided and built Raspian back up to Pixel including all the UI mods.
At this stage everything seemed perfect.
I then installed Chromium browser, rebooted and RaspiWifi no longer worked again.
No AP hotspot. i tried pulling pin 18 high and it did not restart the system.

Any help or guidance would be greatly appreciative 

Dan

jasbur, 2018-09-06:
@SmithtekWA are you starting with the Raspbian Lite image then building up to Raspbian with Pixel, or are you starting with full Raspbian Pixel?

@ngservices It looks like you've done everything right. Based on the comment from @SmithtekWA I'm starting to wonder if there's something specific about full Raspbian that's causing an issue. I usually test with Raspbian Lite. I'm going to re-test the current code with full Raspbian as soon as I get a chance.

Also, I'm going to re-open this issue since it seems like we still have some issues.

SmithtekWA, 2018-09-06:
Hello Jason.

Apologies, I am Smithtek & Ng Services. Sorry for the Bombardment and confusion.

Method install 1
My initial attempts were to use the full Raspian version, "Raspian stretch with desktop"
Direct from Raspberrypi.org.
Following the steps i have outlined above using "Smithtek account", i cannot get the RaspiWifi Webserver to load. 

Method install 2
The other method i used was, i downloaded your pre made image with "Raspbian stretch Lite"
I then build on top of this image, adding the GUI and other mods, essentially turing it back into a full version with desktop. This worked OK. However! i installed Chromium and it had somehow wiped/stopped RaspiWifi working. 

I'm going to do another test on the  method 2 today. During the installation of Chromium i plugged in an Ethernet wire to see if RaspiWifi can change between Wifi and ethernet given the opportunity.
It was when i came back to check the status of the Chromium install i discovered Raspiwifi didn't work. 
Whilst on the topic does RaspiWifi handle using the ethernet at any given time ? when wifi is configured and when wifi is not configured ?

Again sorry for the multiple Git users. 1 is work office and 1 is home computer.

Love the software, amazing work mate :0

Dan

jasbur, 2018-09-07:
No problem on the two accounts, I just thought somebody else was hopping in on the fun with a similar problem.

FYI, The Ethernet port should remain active and fully functional in both modes. I might get a chance tomorrow to test using the configurations you laid out. I'll let you know what I find.

jasbur, 2018-09-10:
Ok, I think I may have the answer and a fix. When the Configuration Server is started it references the `/etc/raspiwifi/raspiwifi.conf` file to determine which port to start on (lines `93-96` in `/usr/lib/raspiwifi/configuration_app/app.py` handle this).

For some unknown reason Raspbian Lite is able to handle this port number presented as a string and Raspbian Full errors out because it's not an integer. Anyway, I just explicitly convert it into an integer now and everything seems to work on both versions.

If you could test it out for me by cloning or downloading the [rasbian_full_server_fix](https://github.com/jasbur/RaspiWiFi/tree/rasbian_full_server_fix) branch I would appreciate it, but it took care of the problem for me (which I did see following your steps, thanks for that).

ngservices, 2018-09-10:
Hello Good Sir.

I will try this again today and get back to you ASAP. Thanks for addressing so quickly.
Do you have a personal Email i can contact you regarding something a bit more private ?.

Mine is daniel@smithtek.com.au

Thanks Jason.

Dan

ngservices, 2018-09-10:
Hello Mate.
Great stuff. It worked OK. i downloaded the Git and executed it from the pi.
AP web server opened efficiently and with no issues. Ill keep playing with it and let you know if there are any other issues.

Thanks again

Dan

jasbur, 2018-09-10:
Excellent! I'll close this issue and merge the changes into the Master branch a little later on. 

Feel free to contact me at jasbur@gmail.com as well.

👍

closed 2018-09-10 by jasbur