jasbur/RaspiWiFi #49: Not redireting to save_credentials.html

scottrpaterson, 2018-09-13:
After entering credentials, it does not redirect to save_credentials.html.

It does however correctly save the credentials and reboot. Tested on Firefox 62 with newest img version on Pi Zero W.

Unrelated question, but in /etc/raspiwifi/raspiwifi.conf does Auto-Config mode check for an "active connection" to the internet via ping or to the local router?

jasbur, 2018-09-19:
Neither, I wanted to make sure to account for situations where a user might have a router set up intentionally with no internet access for local commutation between devices (pinging an internet resource would fail here). Also, I wanted to account for situations where firewall rules and unique network topography that might limit the success of pinging the Pi's default gateway or immediate router.

So, I'm actually using the `wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf` command to attempt a connection, then the `wpa_cli -i wlan0 status` command to query for the connection status. It outputs something like:

```
bssid=xx:xx:xx:xx:xx:xx
freq=5180
ssid=My SSID
id=0
mode=station
pairwise_cipher=CCMP
group_cipher=CCMP
key_mgmt=WPA2-PSK
wpa_state=COMPLETED
ip_address=192.168.1.3
p2p_device_address=xx:xx:xx:xx:xx:xx
address=xx:xx:xx:xx:xx:xx
uuid=xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxx
```

The I just search for `wpa_state=COMPLETED` in the output. If it's there then that means that the full connection process, including the correct key, has been successful.

Also, I think I know what the problem is regarding the `save_credentials.html` no displaying after a successful connection. I'm working on it in the `authentication_check` branch,

jasbur, 2018-10-01:
Thanks @alexrj for the fix.

closed 2018-10-01 by jasbur