jasbur/RaspiWiFi #51: Use a base wpa_supplicant.conf template

alexrj, 2018-09-28:
It would be handy to create a `/etc/wpa_supplicant/wpa_supplicant.conf.base` file that gets copied to `/etc/wpa_supplicant/wpa_supplicant.conf` before the new network is written to it. This way a few known hard-coded networks can be stored anyway, along with any other custom configuration option.

ashah7, 2018-11-08:
I found that not having a wpa_supplicant.conf file at all causes wlan0 to be in a "down" state on reboot.
This happens intermittently.

jasbur, 2019-01-25:
I just released v2.0 wich no longer deletes `/etc/wpa_supplicant/wpa_supplicant.conf` but instead just renames it to `wpa_supplicant.conf.original`

The new Uninstaller script will just copy back the original then which should take care some of these issues.

closed 2019-01-25 by jasbur