jasbur/RaspiWiFi #56: A reset while in AP host mode overwrites needed original files

Aynumi, 2018-10-10:
The reset.py is active during AP Host. 
If the main logic loop waiting for a button is triggered during AP Host  then the function `reset_to_host_mode()` is triggered

This results in a bug where the orginal files are over written by the `reset_to_host_mode()` function.
```
os.system('mv /etc/dhcpcd.conf /etc/dhcpcd.conf.original')
os.system('mv /etc/dnsmasq.conf /etc/dnsmasq.conf.original')
```
After connecting to a Wi-Fi (The original files are now incorrect) an issue arrises where DHCP is set to local and although connected to Wi-FI cant access due to nameserver resolving locally.

To resolve this, I created an entirely separate file (dhcpcd & dnsmasq) and am copying these files during the `set_ap_client_mode()` in rather than depending on the flow to save the original files & replace them.

ashah7, 2018-11-06:
definately and issue here

emsilva, 2018-11-08:
Created pull request #59 to avoid this problem. Its a dirt fix, but works.