jasbur/RaspiWiFi #57: DNS when on client mode

emsilva, 2018-11-05:
My DNS is defaulting to 127.0.0.1, instead of using the DNS provided by the DHCP client on my network. I tried different routers and getting the same results. I'm using the latest img release.

`
# Generated by resolvconf
nameserver 127.0.0.1
`

emsilva, 2018-11-06:
Stupid that the 127.0.0.1 is how things works... I'm still having DNS problems, but I'll find out better what is the problem before I post again.

closed 2018-11-06 by emsilva

maxpage92, 2019-10-03:
Did you ever figure it out? I'm having the same issue where resolv.conf is 127.0.0.1 and it's causing some networking issues. I have my pi running as a kiosk on boot and for the first 10 seconds after chromium loads it has a dns error. Then it loads correctly. Ping works well beforehand so it must be this. Any solutions?