jasbur/RaspiWiFi #60: manual_reset erroring out

pablobolomey, 2018-11-26:
Hi try to run manual reset and getting following error:

SyntaxError: invalid syntax
pi@raspberrypi:/usr/lib/raspiwifi/reset_device $ sudo python manual_reset.py
Traceback (most recent call last):
  File "manual_reset.py", line 1, in <module>
    import reset_lib
  File "/usr/lib/raspiwifi/reset_device/reset_lib.py", line 31
    print(line.replace("temp-ssid", ssid_prefix + serial_last_four), end='')

pablobolomey, 2018-11-26:
switch to master clone download and not release file

pablobolomey, 2018-11-28:
issue is also on master build

jasbur, 2018-11-28:
I just ran the manual reset successfully on a fresh install of Raspbian with the current master branch and didn't have any issues.

I think I might have just spotted your problem though. You're calling the command like this:

`sudo python manual_reset.py`

But raspbian still devault's to Python 2, so you need to explicitly call Python 3 by issuing the command this way:

`sudo python3 manual_reset.py`

jasbur, 2018-12-05:
Since I haven't heard back I'm going to assume that this was a problem with using `python` instead of `python3` when calling the reset function and close the issue.

closed 2018-12-05 by jasbur