jasbur/RaspiWiFi #61: swapping build onto another raspi doesnt change last 4 serial on ssid

ashah7, 2018-11-26:
Seems like this is set only once during the installation.
If the same build is used on another raspi then the last 4 serial numbers on the ssid are not adjusted for
Perhaps this should be checked/adjusted on each boot

jasbur, 2018-12-12:
That's a good idea. I'll toss it into the mixed for future development. 

In the meantime, if you transfer an image to a new device you can manually edit the `ssid=` in `/etc/hostapd/hostapd.conf` to read `ssid=temp-ssid`. When the `libs\reset_device\reset.py`  sees `ssid=temp-ssid` on the next reboot it will change the SSID based on the current hardware ID. 

You can do that as many times as you want, the only dteriment is a slightly longer boot process the first time after changing it.

RedSparr0w, 2019-06-18:
Would changing [this line](https://github.com/jasbur/RaspiWiFi/blob/593531c8a662509d0ced89513101a36df8f348cb/libs/reset_device/reset_lib.py#L53) to check for `ssid_prefix` & `serial_last_four` fix this issue:

```python
			if ssid_prefix in line:
				if serial_last_four in line:
					ssid_correct = True
```

https://github.com/jasbur/RaspiWiFi/blob/593531c8a662509d0ced89513101a36df8f348cb/libs/reset_device/reset_lib.py#L53