jasbur/RaspiWiFi #63: Problem with reinstalling

m-kru, 2018-12-12:
I installed RaspiWiFi and I could see WiFi network to configure my Raspberry PI but I was not able to get to the page 10.0.0.1. I wanted to reinstall RaspiWiFi but without SSL support so I have removed `/etc/raspiwifi/`,  and `/usr/lib/raspiwifi/`. Then I have reinstalled it by executing `sudo python3 initial_setup.py` however this time I do not even see WiFi network that should be created by Raspberry Pi. All configuration options are set to default. Do you know what might be wrong?

jasbur, 2018-12-12:
Unfortunately, by mainly removing those directories and re-running the setup I think the installation is pretty broken at this point. The code doesn't really expect for that to happen. You're probably going to have to reimage your base installation To get it working again. 

For future reference, you can just edit the `ssl_enabled=1` line in `/etc/raspiwifi/raspiwifi.conf` to `ssl_enabled=0` and reboot the device to change the SSL behavior.

closed 2018-12-12 by jasbur

jasbur, 2019-01-20:
I just added an Uninstaller to the project. If you update your repository or re-clone it you can run the Uninstaller like this:

`sudo python3 libs/uninstall.py`

Install from now on can always access it from `/usr/lib/raspiwifi/uninstall.py` as well.

This will help by removing any configuration relating to RaspiWiFi.