jasbur/RaspiWiFi #64: Doesn't work on Raspberry Pi 3 Model B

proudloyaltier, 2018-12-12:
The connection page never loads and does not open config mode on startup

DustinDobernig, 2019-01-04:
Experiencing same issue

jasbur, 2019-01-04:
Can either of you try commenting out or removing `Line 2` (`driver=nl80211`) in `libs/reset_device/static_files/hostapd.conf` (or in `/etc/hostapd/hostapd.conf` if you've already run the `initial_setup.py`) to see if that makes a difference?

proudloyaltier, 2019-01-04:
I tried reinstalling RaspiWiFi, and it worked perfectly fine!!  I tried it on 2 other Pi 3B and it worked first try on one and second try on the other

jasbur, 2019-01-25:
I just tested on a Raspberry Pi 3 B+ with no issues. I'll close this for now and chalk it up to fluke until we have more consistent information to work from.

closed 2019-01-25 by jasbur