jasbur/RaspiWiFi #66: cant see Access Point after initial setup

fu-techiot, 2018-12-18:
i'm using raspberry Pi Zero, I have this ZIp in Downloads after initial setup done. i can't find access point of raspberry Pi zero w.
in  rasspiwifi.conf i have following credential strode base on initial setup 
ssid_prefix = ankit
auto_config=1
auto_config_delay=240
ssl_enabled = 0
server_port = 80

jasbur, 2019-01-25:
What kind of output do you get if you run `sudo hostapd -dd /etc/hostapd/hostapd.conf` from the Pi?

augisbud, 2019-02-26:
```
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
Could not open configuration file '/etc/hostapd/hostapd.conf' for reading.
Failed to set up interface with /etc/hostapd/hostapd.conf
hostapd_init: free iface 0x18adcd8
Failed to initialize interface

```
I have the same issue, this is what I get

jasbur, 2019-02-26:
Thanks for the information. I'll look into it as soon as I get the chance.

ashah7, 2019-02-26:
unfortunately this has happened to me before
A rebuild was the only thing that fixed it for me

bartbergman, 2019-07-12:
I had the same issue.
For me it was caused by a too short WPA key (needs to be 8 - 63 characters long). This crashed hostapd.
The reason for failing is mentioned very clearly in the output of `sudo hostapd -dd /etc/hostapd/hostapd.conf` by the way.

vicwomg, 2019-12-28:
First of all, this is a great project. Works awesome and solves the problem I was working on.

But, I ran in to this initial issue too. For me, it was due to setting the WPA password to blank after setting WPA to on in the initial setup. These two are mutually exclusive. My intention was to disable WPA.

In my case, to fix this, you'll need to log in and edit your /etc/raspiwifi/raspiwifi.conf :

```
wpa_enabled=0
wpa_key=
```

As a feature request, the initial setup UI should not accept invalid WPA keys (must be 8-63 chars) to prevent this condition.