jasbur/RaspiWiFi #69: All networks not found when finding Wifi Networks

Denness, 2019-01-08:
After installing RaspiWiF, I was able to connect and see the page at 10.0.0.1. When attempting to select a wifi network less than half of the wifi networks in the building were found.

I work in a serviced office and there are over 20 networks that I can usually join from the Pi Zero W. However, after installing RaspiWiFi there are only about 10 available networks that I can join and our network is not on the list. 

I attempted to manually enter our wifi network ssid and password which caused the device to reset. After the reset the pi was unable to join the network.

By looking at the networks in the buildings I then found that the RaspiWifi is only picking up Wifi networks on channels 1-11. But my laptop is able to find networks that are on channels 1-100.

I hope this information helps find a solution as RaspiWifi seems like a great tool.

jasbur, 2019-01-08:
I know, in the US, 2 Ghz wireless networks are only allowed to broadcast on channels 1-11. 5 Ghz networks generally broadcasting the channels over 100. I know there are some differences in the Wi-Fi specification depending on region though. What region are you in?

Denness, 2019-01-08:
I'm in the UK. I didn't know about these differences.

closed 2019-01-25 by jasbur