jasbur/RaspiWiFi #75: config page url

rboorman, 2019-06-17:
This is more of an enhancement but for me a really crucial one.   Is there a way we can have the configuration page goto a configuration URI path - like http://whatever.com/<URI> e.g. /configure 

I am putting together an opensource pi image with our software that we want to be on port 80 and at the root.

I really like the way you have things setup and appreciate what you have created here.

Thank you

Robert

RedSparr0w, 2019-06-17:
Just edit the config file there is already a custom URI `raspiwifisetup.com` which will show the config page, just add your own, or change that one.

https://github.com/jasbur/RaspiWiFi/blob/a7ca3f1715901fe0c44c76f889704aead5c6a5b5/libs/reset_device/static_files/dnsmasq.conf#L10

There is also `idliketoconfigurethewifionthisdevicenowplease.com` as another URI.

rboorman, 2019-06-17:
Oh Yeah - I use that already - i was hoping to have it be hosted at /config not at the root - I have a web server I am loading up for my app on port 80 - could I create a route to your page?

Thank you for responding

rboorman, 2019-06-18:
I think this would do the initial thing I want:

@app.route('/config')

I didn't even thing to look into what web server was being used.  I'll try https://www.npmjs.com/package/flask-node - to incorporate your page into my node app.

Anything you point out that may be a gotcha?

Thank you

Robert

RedSparr0w, 2019-06-18:
Ah i see,

This line confused me:
>I am putting together an opensource pi image with our software that we want to be on port 80 and at the root.

Depending what you are using to host your app, you should be able to make a redirect to 10.0.0.1 maybe?

Or use an iframe: _(Not tested, but worth a try)_
```html
<iframe src="10.0.0.1" style="position:fixed; top:0; left:0; bottom:0; right:0; width:100%; height:100%; border:none; margin:0; padding:0; overflow:hidden; z-index:999999;">
    Your browser doesn't support iframes
</iframe>
```

rboorman, 2019-06-18:
The iframe actually helps with the next step of what i'm trying to accomplish but for now - i thin  @app.route('/config') will do the trick.

Thank you!

RedSparr0w, 2019-06-18:
No worries, if you do come up with another solution, you should post it here as it may help others in the future :smile:

rboorman, 2019-06-18:
Absolutely will!

Dimasua, 2021-07-10:
Hej ReadSparrOw, really appreciate your work for the script you created. I have successfully connected my Raspi to my home wifi with it. However I could not figure out what is a web page with the main configuration? I checked my Raspi IP via home router, but it is not accessible over a browser on this IP. Is there anything I might be missing?