jasbur/RaspiWiFi #78: Please update for Buster / Raspberry Pi 4

jbrepogmailcom, 2019-07-28:
Hello,

I was not able to install so I want to try the image. However, I use Raspberry Pi 4 and I think that Raspbian Stretch image that can be downloaded in Releases does not contain drivers for latest RasPi 4 hardware like USB 3.0.

Would it be possible to create .IMG for latest Raspbian Buster so it can be used for RasPi 4?

Thank you

jbrepogmailcom, 2019-07-28:
I have tested the image for Raspberry Pi 4 and it does not boot.

doronhorwitz, 2019-08-02:
I can confirm that RaspiWiFi stills works for me under Raspian with a RaspberryPi 3 B+.
I did a clean installation of Raspian and then installed RaspiWifi as per the Readme.txt
Installation worked exactly according to the instructions - had to do nothing unexpected.

tonyfdesign, 2019-09-11:
Working fine for me on a Buster install.

invensAB, 2019-11-28:
Works for me on Buster but when it replaces the wpa_supplient.conf is misses to add the country line.
country=US

nmstoker, 2021-10-03:
@jbrepogmailcom - I realise it's a long time since you flagged this, but was interested to know if you resolved it?

One thing worth trying is to see if the script based instructions work for you on a Pi 4.  It seems likely that the image would indeed not work as I believe there are some changed the Pi 4 introduced but my guess is that the script approach would bypass these since you'd be using a base image that works on the Pi 4 and that should side-step  the fact the old image here isn't Pi 4 compatibile.