jasbur/RaspiWiFi #79: resolve all hosts to 10.0.0.1 in host_mode

jgruber, 2019-07-30:
Current dnsmasq resolves only two special hostnames to the default 10.0.0.1 address. A simple configuration change to the static_files/dnsmasq.conf file with this entry:

`address=/#/10.0.0.1
`

would resolve all hosts to 10.0.0.1

Additionally, if Flash is configured to have a self-signed certificate with a TLS listener redirecting you to http://10.0.0.1, it would provide a familiar 'sign-in' captive portal like experience typical hotels or other IoT gateways.

RedSparr0w, 2019-09-06:
Would be awesome to have this optionally enabled

tonyfdesign, 2019-09-26:
On my phone any URL will arrive at 10.0.0.1—works as described. On desktop, any URL will find the internet. Only 10.0.0.1 displays the RaspiWifi information.

mwargan, 2019-10-02:
@jgruber good tip on the `address`! What do you mean by "Flash is configured to have a self-signed certificate with a TLS listener redirecting you to http://10.0.0.1"? How is Flash used in the python code, and how can I go about setting this up?