jasbur/RaspiWiFi #88: ssid is not displaying last 4 serial

ashah7, 2019-11-28:
Previous versions worked fine for me, I've been using it for over a year with Raspian Lite Stretch

Using Buster Lite - fresh install and the last 4 of the serial in hostapd.conf are incorrect 
below is what it shows:
ssid should be "RaspiWifi c873"

pi@raspberrypi:~ $ cat /proc/cpuinfo | grep Serial | awk '{print $3}'
00000000321ec873

cat /etc/hostapd/hostapd.conf
interface=wlan0
driver=nl80211
ssid=RaspiWifi 1.2  <--incorrect not sure where the 1.2 comes from
channel=1
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=getmeyellow

corrilan, 2021-08-31:
Hello

I am also seeing this issue using Buster on Pi Zero W.

root@photoframe-lounge-the-mount:~# cat /proc/cpuinfo | grep Serial | awk '{print $3}'
00000000ee700f02

The generated access point name is "photo-frame- 1.1" whereas I was expecting "photo-frame-0f02".

Regards

Jason

BramWerbrouck, 2021-09-02:
> Hello
> 
> I am also seeing this issue using Buster on Pi Zero W.
> 
> root@photoframe-lounge-the-mount:~# cat /proc/cpuinfo | grep Serial | awk '{print $3}'
> 00000000ee700f02
> 
> The generated access point name is "photo-frame- 1.1" whereas I was expecting "photo-frame-0f02".
> 
> Regards
> 
> Jason

Same here: myNetwork 1.1