jasbur/RaspiWiFi #92: Address remain static (10.0.0.1) in client mode

Zefram88, 2020-03-18:
I'm able to solve it simply removing dhcpcd.conf, but that is needed to be done every time host mode is invoked.

rklass, 2020-05-06:
I have the same problem, but I could not solve it. If I remove dhcpcd.conf and press again button to start raspiwifi again, it is not possible to connect

beaconIOT, 2020-07-04:
I have the same issue. Any fix? After it reboots and connects to my wifi, it still shows as the static IP: 

MacBook-Pro:~ Kris$ ping raspberrypi.local
PING raspberrypi.local (10.0.0.1): 56 data bytes

trailhead, 2020-11-05:
I'm looking at this now and I suspect it was caused on my system by running manual_reset.py without root permissions.  You end up in a really broken state where both host and client modes are trying to bootstrap at the same time, and you still have the host mode dhcpcd.conf and IP 10.0.0.1

nishmeht7, 2021-03-02:
I think this issue is being caused by multiple entries of this in `etc/crontab`

```
'echo "@reboot root run-parts /etc/cron.raspiwifi/" >> /etc/crontab'
```

Every time I run reset, the setup_lib.py runs and adds this line to `/etc/crontab` and this works the first time since it doesn't already exist, but when this runs a second time and creates a duplicate entry for the host mode stops working.

Removing the duplicate line and rebooting should successfully get the pi back to host_mode.