jasbur/RaspiWiFi #96: Raspberry Pi4 support?

djlongy, 2020-06-19:
Tried this on my pi4 running latest retropie and it doesn't seem to work after a successful setup.
Waiting 300 seconds doesn't trigger autoconfig mode (enabled)
Not sure if it's just retropie?

Wheaties466, 2020-10-19:
I also noticed that the pi 4 isn't supported. @jasbur If you want to hand over the project to someone else I would be willing to assist.

JohnDStrand, 2020-12-23:
I installed it on a Pi 4, seems to work, but I can't access any service ports other than the screen to set your WiFi credentials.

BramWerbrouck, 2021-08-29:
Is there any update on how to get this properly working on a Raspberry Pi model 4?

JohnDStrand, 2021-08-29:
I was able to open the other ports by setting the port number manually in the initial set-up to another value than 80, the default. It has been working great for me. If you have any specific questions, please feel free to email me.

BramWerbrouck, 2021-08-29:
> If you have any specific questions, please feel free to email me.

can you give me your email address?

JohnDStrand, 2021-08-29:
Here you go!

BramWerbrouck, 2021-08-29:
> Here you go!

am I looking at the wrong place? (I don’t see your email address)

JohnDStrand, 2021-08-29:
sorry, I didn't realize how this worked. but, to be safe, I don't want to post my email on github. Maybe you can just post your questions here?

BramWerbrouck, 2021-08-29:
> sorry, I didn't realize how this worked. but, to be safe, I don't want to post my email on github. Maybe you can just post your questions here?

I’ll give you mine: bram.werbrouck@gmail.com

samuk, 2021-10-15:
Running script on a new focal install. Wifi network is created but no page is visible on 10.0.0.1

feltz916, 2022-03-11:
This is almost working on my Pi4, however i am stuck.

I used the suggestion above by changing the default port to something other than 80, and that works.
I am able to see the AP, set a new network, put in password, and apply settings. Once it resets however, my device is still stuck on 10.0.0.1 static IP address and does not join the specified network.