jasbur/RaspiWiFi #118: Simplest way to detect if running in host mode or not?

benwiley4000, 2022-09-21:
I'm trying to configure another program to know when RaspiWiFi is in host/configuration mode or just running normally. Is there an easy way to do this?

jediempi, 2022-09-21:
I use the IP address to check if it is in host/configuration mode. This is my easy script... :)

`#!/bin/bash`
case $(ifconfig)
in 
  *"inet 10.0.0.1"*) 
     echo $inet "IP found in 10 range for AP mode, leaving DNS alone" 
   ;;
  *) 
    echo $inet "Not in AP mode anymore, let's call the next script" 
    /bin/bash /home/pi/wifi_fix_dns.sh
   ;; 
esac`

jediempi, 2022-09-21:
[image]

benwiley4000, 2022-09-22:
Thank you! Super helpful.