jasbur/RaspiWiFi #123: Does not work with Bookworm

mdskier, 2024-03-14:
Tried to use this with a new bookworm install, but not having success.

 Bookworm does not use wpa_supplicant,conf, so that is likely part of the problem. Starting to look at details and see if I am able to make changes required to use .nmconnection files instead.

Sumlayen, 2024-07-17:
Following this as I have the same issue.

rogersmj, 2025-02-12:
Did you get anywhere on this? I've been trying to figure out why it wasn't working...it's because I'm on Bookworm.

bauer-san, 2025-02-24:
@rogersmj have you tried something like this: https://github.com/bauer-san/CaptivePortal on Bookworm

NeonLightning, 2025-03-24:
honestly i found it easier to just install hostapd and dnsmasq and make my own python script using flask nmcli and iw. i couldn't get captiveportal to work